microsoft/ELL #203: Wrap.py error

herodangau, 2019-03-12:
Hi, I have been trying to follow the first tutorial (https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/) on my windows pc.
When i tried to run the Wrap.py module, it just stopped with the below message which is different from the tutorial.

(py36) C:\Users\dtlabvr\ELL>python C:/Users/dtlabvr/ELL/tools/wrap/wrap.py --model_file model.ell --language python --target host
copy "C:\Users\dtlabvr\ELL\CMake/OpenBLASSetup.cmake" "host\OpenBLASSetup.cmake"
compiling model...
C:/Users/dtlabvr/ELL/build/bin/release/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas false --optimize true
### WrapException: <class 'buildtools.EllBuildToolsRunException'>: C:/Users/dtlabvr/ELL/build/bin/release/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas false --optimize true

lovettchris, 2019-03-12:
Can you run just the compile command line by itself and report the output you see?
```
C:/Users/dtlabvr/ELL/build/bin/release/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas false --optimize true
```

lovettchris, 2020-12-02:
closing due to inactivity, but if you want further investigation please reopen and add your host CPU details so we can debug this.

closed 2020-12-02 by lovettchris