microsoft/ELL #204: create a cmake to run example

FoxMarts, 2019-03-14:
Hi, I'm new to this library, I'm trying to run the existing ELL/libraries/trainers/test/src folder in another directory outside of ELL, but I'm having trouble creating CMakeLists or simply creating a Makefile to create the executable . Could anyone give an example of a CMakeLists.txt file.
Best regards.

lovettchris, 2019-03-14:
Just follow the setup and build instructions that match your operating system:

* [Windows](https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md)
* [Ubuntu Linux](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md)
* [Mac OS X](https://github.com/Microsoft/ELL/blob/master/INSTALL-Mac.md)

when you are done the above test app will appear in the build/libraries/trainers folder.

FoxMarts, 2019-03-18:
After a few tries, I was able to make a cmake that just runs the example outside of the ELL directory.
Now I want to change the code to save the SDGTrainer Model in the same folder as the executable, but i do not know what new headers to put in the code and what to change or change in cmake.
Can someone help?
Best regards.
The CMakefile 

cmake_minimum_required(VERSION 3.8 FATAL_ERROR)
#
# cmake file for forestTrainer project
#
project(ELLtest)

#ELL location 
set(ELL ELL/)

#ELL libraries location
set(ELL_LIBRARIES_DIR ${ELL}/libraries)

#ELL link library

set(ELL_LINK_DIR ${ELL}/build/libraries/)

# Include modules in the CMake directory.
list(APPEND CMAKE_MODULE_PATH "${ELL}/CMake")
include(CompilerCache)

# Set C++ version
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

# Set GCC-specific options
add_compile_options(-Wall)
add_compile_options(-Wsign-compare)
add_compile_options(-Wno-missing-braces)
add_compile_options(-Wmissing-field-initializers)
add_compile_options(-fvisibility-inlines-hidden)
add_compile_options(-Wno-unknown-pragmas)
set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG} -ggdb3 -O0")
set(CMAKE_C_FLAGS_DEBUG "${CMAKE_C_FLAGS_DEBUG} -ggdb3 -O0")
set(CMAKE_CXX_FLAGS_RELWITHDEBINFO "${CMAKE_CXX_FLAGS_RELWITHDEBINFO} -ggdb3")
set(CMAKE_C_FLAGS_RELWITHDEBINFO "${CMAKE_C_FLAGS_RELWITHDEBINFO} -ggdb3")

# Set up global variables to help find NuGet projects
#set(PACKAGE_ROOT ${EXTERNAL_DIR})
include(OpenBLASSetup)
include(LLVMSetup)
include(SWIGSetup)
include(CopySharedLibraries)
include(AddPrecompiledHeader)

find_library(trainers trainers ${ELL_LINK_DIR}/trainers)
find_library(functions functions ${ELL_LINK_DIR}/functions)
find_library(testing testing ${ELL_LINK_DIR}/testing)
find_library(utilities utilities ${ELL_LINK_DIR}/utilities)
find_library(data data ${ELL_LINK_DIR}/data)

# define project
set (test_name test)

set (test_src main.cpp)

source_group("src" FILES ${test_src})

add_executable(${test_name} ${test_src})
target_include_directories(${test_name} PRIVATE include ${ELL_LIBRARIES_DIR})

target_link_libraries(
	${test_name}
	${trainers} 
	${functions} 
	${testing} 
	${utilities}
	${data}
)

closed 2019-05-23 by FoxMarts

lovettchris, 2019-05-23:
thanks, yes we should make this easier.

arachid1, 2020-06-03:
@lovettchris do your folders need to reside outside of the ELL directory? For example, when completing the tutorials, does your tutorial folder need to be created outside of ELL? That wasn't clear from the tutorials. 

Thank you!