microsoft/ELL #205: compile tool error

FoxMarts, 2019-03-19:
Hi, I created a ProtoNN model using the command:

/ELL/build/bin/protoNNTrainer -idf protonnTestData.txt -dd 20 -omf testProtonn

But when I try to create a C header for use in the nRF52 Development Kit, which contains the cortex m4 arm, using the given compile example:

/ELL/build/bin/compile -imf testProtonn -cfn Predict -cmn model --bitcode --fuseLinearOps True --header --blas false --optimize true --target custom --numBits 32 --cpu cortex-m4 --triple armv6m-gnueabi --features +vfp4,+d16,+soft-float

A error appears 
exception: Failed to match field nodes, instead found token 'model'

if you change the command by replacing -imf for the -imap

The error is:
compile: ELL/libraries/emitters/src/IRModuleEmitter.cpp:885: const llvm::DataLayout& ell::emitters::IRModuleEmitter::GetTargetDataLayout() const: Assertion `!GetLLVMModule()->getDataLayout().getStringRepresentation().empty()' failed.
Aborted (core dumped)

Best Regards.

lovettchris, 2019-03-19:
Thanks for the bug report.  I can reproduce this bug.  We'll get it fixed.

lovettchris, 2019-04-29:
This works properly now if you use the latest release 2.5.0

closed 2019-04-29 by lovettchris