microsoft/ELL #207: Error building on MacOS 10.14.3 Mojave

maniflames, 2019-03-26:
Hi all, 

First of all thanks for creating this, I can't wait to try it out! 
Right now I'm following the installation guide for installing ELL on MacOS but I ran into some issues. Right now I'm stuck at a generic 'Undefined symbols for architecture x86_x64' error after I run make. It looks something like this: 

```
[ 47%] Linking CXX static library libcommon.a
[ 47%] Built target common
Scanning dependencies of target makeExamples
[ 47%] Building CXX object tools/utilities/makeExamples/CMakeFiles/makeExamples.dir/src/main.cpp.o
[ 47%] Building CXX object tools/utilities/makeExamples/CMakeFiles/makeExamples.dir/src/GenerateModels.cpp.o
[ 48%] Building CXX object tools/utilities/makeExamples/CMakeFiles/makeExamples.dir/src/ModelGenerateArguments.cpp.o
[ 48%] Linking CXX executable ../../../bin/makeExamples
Undefined symbols for architecture x86_64:
  "ell::math::Blas::Dot(int, double const*, int, double const*, int)", referenced from:
      __ZZZN3ell5value3DotENS0_6VectorES1_ENK3$_3clINS0_14ComputeContextEEEDaRT_ENKUlNS0_6ScalarES1_S8_S1_S8_E_clES8_S1_S8_S1_S8_ in libvalue.a(VectorOperations.cpp.o)
  "ell::math::Blas::Dot(int, float const*, int, float const*, int)", referenced from:
      __ZZZN3ell5value3DotENS0_6VectorES1_ENK3$_2clINS0_14ComputeContextEEEDaRT_ENKUlNS0_6ScalarES1_S8_S1_S8_E_clES8_S1_S8_S1_S8_ in libvalue.a(VectorOperations.cpp.o)
  "std::runtime_error::what() const", referenced from:
      vtable for ell::utilities::CommandLineParserErrorException in libutilities.a(CommandLineParser.cpp.o)
      vtable for ell::utilities::CommandLineParserException in libutilities.a(CommandLineParser.cpp.o)
      vtable for ell::utilities::CommandLineParserPrintHelpException in libutilities.a(CommandLineParser.cpp.o)
      vtable for ell::utilities::CommandLineParserInvalidOptionsException in libutilities.a(CommandLineParser.cpp.o)
... (a really really really long stack)
Dwarf Exception Unwind Info (__eh_frame) in main.cpp.o
      ...
ld: symbol(s) not found for architecture x86_64
clang-6.0: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/makeExamples] Error 1
make[1]: *** [tools/utilities/makeExamples/CMakeFiles/makeExamples.dir/all] Error 2
make: *** [all] Error 2
```
I've noticed that when I run `cmake ..` that it would detect the Clang version that came preinstalled on my mac instead of the one that shipped with LLVM 6 so I started over but ran `CCX=/usr/local/opt/llvm@6/bin/clang CC=/usr/local/opt/llvm@6/bin/clang cmake ..`. Unfortunately it didn't have a lot of impact. I did notice a warning in as well I'm not sure how relevant it is but it looks like this: 

```
...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- BLAS library not found
-- Found LLVM 6.0.1
-- Using LLVMConfig.cmake in: /usr/local/opt/llvm@6/lib/cmake/llvm
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12") 
-- Found SWIG_EXECUTABLE=/usr/local/bin/swig
-- Skipping pitest because one of RPI_CLUSTER, RPI_PASSWORD, RPI_KEY, TEST_MODELS_REPO is missing
-- Creating wrappers for python
CMake Deprecation Warning at /usr/local/share/cmake-3.8/Modules/UseSWIG.cmake:226 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  CMake/CommonInterfaces.cmake:129 (swig_add_module)
  CMake/CommonInterfaces.cmake:160 (generate_interface_module)
  interfaces/python/CMakeLists.txt:21 (generate_interface)
...
```
 
I don't have experience with building c stuff so I'm sorry in advance for missing something obvious. Is anyone able to help me a little with the install?

lovettchris, 2019-03-26:
When changing cmake command line it is best to delete the 'build' folder and start over.  cmake caches a lot of stuff and can get confused if you don't start from scratch.  Can you also provide more information about the exact type and model of machine you are running this on ?

kernhanda, 2019-03-27:
Looks like you don't have a BLAS library available but we're referencing it anyway. This might be our bug.

maniflames, 2019-03-27:
Thanks for your help. I'm working on a MacBook Pro 15-inch (2016) with a 2,6 GHz Intel Core i7. When retrying to build I did remove the entire folder and make a new one. I've installed openblas through homebrew but noticed that it refuses to link even when I run `brew link --force openblas`. 
The message after installation of openblas: 

```
openblas is keg-only, which means it was not symlinked into /usr/local,
because macOS provides BLAS and LAPACK in the Accelerate framework.

For compilers to find openblas you may need to set:
  export LDFLAGS="-L/usr/local/opt/openblas/lib"
  export CPPFLAGS="-I/usr/local/opt/openblas/include"

For pkg-config to find openblas you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/openblas/lib/pkgconfig"
```

I tried to run the commands above as well but it seems to have no effect on anything. How do I make sure either openblas or BLAS provided by apple is linked in the right way?

maniflames, 2019-03-27:
While figuring the BLAS thing out I noticed a couple of things: 
- Homebrew ships with openblas 0.3.5 and the `versions` command was depreciated. (Tried to get the right version from git but failed building that as well)
- In OpenBLASSetup.cmake searches for where a BLAS lib should be availible but the `headers` dirs in the Accelerate Framework were missing for me. I got them by installing the xcode IDE. 

Thanks for your help everyone 😄

closed 2019-03-27 by maniflames

lovettchris, 2019-03-27:
Cool, if there's something we can fix in our OpenBLASSetup.cmake please post a pull request, thanks.