microsoft/ELL #208: _model.pyd file not found while "Getting-started-with-image-classification-on-the-Raspberry-Pi"

VarshaSLalapura, 2019-03-29:


VarshaSLalapura, 2019-03-29:
Hi ELL team,

Am trying to run the tutorial Getting started with image-classification on RPi.
[1-cmake.txt](https://github.com/Microsoft/ELL/files/3021252/1-cmake.txt)
[2-make.txt](https://github.com/Microsoft/ELL/files/3021254/2-make.txt)
After building the ELL, when I go to the tutorial to build a model, model is 
not imported. There is no Release folder containing the _model.pyd file.

Thanks in advance,
Varsha

kernhanda, 2019-03-29:
It seems that you're on a POSIX machine (Linux?), in which case there won't be a Release folder. 

Can you try running the following commands:
```shell
cd /home/varshalalapura/Desktop/git-prjoects/ELL/ELL/build
find . -name '*model.so'
```
That might show you the location of the built model.

VarshaSLalapura, 2019-03-29:
Yes, I could find it. I got the confidence score now. The thing is am novice to linux 
as well as building using GCC. I decided to ask after many trials and you could 
track my stuck point. Thank you so much for all your time.

closed 2019-03-29 by VarshaSLalapura