microsoft/ELL #209: Unable to compile latest featurizes

maniflames, 2019-03-31:
Hi all, 

After installing ELL I ran into an error while trying to compile the latest Cinnamon featurizer. I've tried other flavours as well but ran into the same problem. 

When I run `python3 ../ELL/tools/wrap/wrap.py --model_file featurizer.ell --target host --outdir featurizer_16k --module_name mfcc`

I get the following output: 
```
### WrapException: <class 'buildtools.EllBuildToolsRunException'>: /Users/maniflames/Documents/Development/cmgt/robot/experiments/ELL/build/bin/compile -imap featurizer.ell -cfn Predict -cmn mfcc --bitcode --target host -od featurizer_16k --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas true --optimize true
```

Running that command prints the following error:
```
exception: Error: reading ELL map: featurizer.ell: Failed to match token }, got: "
```

I tried to find the mismatch manually but the file pretty large and I missed it. 
I am able to successfully compile and build the classifiers. I'm also able to compile and build both the featurizers and classifiers from commit [fcfb666](https://github.com/Microsoft/ELL-models/tree/fcfb666d377baee3eca82a226537152d07b5a5e1) of the models repo. 

My setup & environment: 

```
OS: MacOS 10.14.3 (Mojave)
Processor: 2,6 GHz Intel Core i7
Clang: version 6.0.1
BLAS library: libBLAS in MacOS Accelerate framework
```

I hope you can help out, would love to try the new models.
Thanks again for all the work on ELL.

lovettchris, 2019-04-01:
Yeah, sorry about that the speech commands models and the ELL repo are temporarily out of sync.  Will be fixed shortly, in the meantime stick with fcfb666 for now.

lovettchris, 2019-04-29:
The latest release brings the ell-models and ell repos back in sync, please move to version 2.5.0

closed 2019-04-29 by lovettchris