microsoft/ELL #211: python tutorial breaks

yunqu, 2019-04-02:
I have followed the tutorial: https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/

At the step of running `python /opt/builds/ELL/tools/wrap/wrap.py --model_file model.ell --language python --target host`, I was stuck at the LLVM step, which is relevant to #195 and #196 .  

I proceeded for building raspberry pi part and it works perfectly on my aarch64 (ultra96 board). One note is that I need to change the compiler flag in my ELL repo:

```
find ./ -type f -exec sed -i 's/-mtriple=armv7-linux-gnueabihf/-mtriple=aarch64-unknown-linux-gnu/gI' {} \;
```

This makes me happy.

So I guess my only concern is that the host side part verification cannot be done in the current flow.

lovettchris, 2019-04-03:
did you try "--target aarch64" ?

lovettchris, 2020-12-02:
closing due to inactivity.  If you want to investigate this further we'll need to know more about the CPU in your host computer.

closed 2020-12-02 by lovettchris