microsoft/ELL #212: OpenCV packages path

Ry7en, 2019-04-10:
Having followed the steps necessary to build python 3.5 on my pi zero, I noticed in the CMake configuration summary that my "packages path" differed from the one shown in the tutorial:

```
--   Python 3:
--     Interpreter:                 /usr/bin/python3 (ver 3.5.3)
--     Libraries:                   /usr/lib/arm-linux-gnueabihf/libpython3.5m.so (ver 3.5.3)
--     numpy:                       /usr/lib/python3/dist-packages/numpy/core/include (ver 1.12.1)
--     packages path:               lib/python3.5/dist-packages

```
Choosing to finish the installation, I found that

 `python3 -c "import cv2; print(cv2.__version__)"` 

correctly answered 3.3.0, but running the tutorial.py script as required in the "Getting started with image classification on Raspberry Pi" tutorial outputted:

```
Traceback (most recent call last):
  File "tutorial.py", line 13, in <module>
    import cv2
ImportError: No module named cv2
```
If I change the command to run the script from "python tutorial.py" to "python3 tutorial.py" the error differs:

```
Traceback (most recent call last):
  File "tutorial.py", line 19, in <module>
    import model
ImportError: No module named 'model'
```

Having searched the pi, a"dist-packages" folder containing "cv2.cpython-35m-arm-linux-gnueabihf.so" is located in "/usr/local/lib/python3.5/dist-packages", and not "/usr/lib/python3.5".

lovettchris, 2019-04-10:
Our normal instructions for [setting Raspberry Pi use Python 3.4[https://microsoft.github.io/ELL/tutorials/Raspberry-Pi-setup/).  But it sounds like you have chosen to follow the [wiki on Python 3.5](https://github.com/Microsoft/ELL/wiki/Building-OpenCV-for-Python-3.5) instead.  These are instructions are for building OpenCV on Python 3.5, not for "building python 3.5".  Python 3.5 comes pre-installed on the latest [Raspberry Pi Rasbian images](https://www.raspberrypi.org/downloads/raspbian/).    

Given that, yes you will use "python3" to invoke Python 3.5 interpreter and it is this version of python that has OpenCV installed, and it looks like your "import cv2" on python3 is working, so congrats - you now know how to build OpenCV from scratch on a raspberry pi.  Cool.

Now for using python3 to run compiled ELL models.  Let's assume you are doing our first tutorial [Getting started with image classification on Raspberry Pi](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/), then in that tutorial, wherever you see the word "python" you need to replace that with "python3".  You also need to skip all the instructions on "miniconda", you are not using a "virtual python environment", instead you are using the system provided python3, so do the tutorial from a clean terminal where you have not run miniconda activation "source activate py34".  Note that miniconda setup may have placed this line in your .bashrc, if so comment it out:
```
export PATH="/home/pi/miniconda3/bin:$PATH"
```
You want `which python3`  to return `/usr/bin/python3`.  Then the wrap.py command should build the right version of the ELL python wrappers that will work properly with python 3.5 interpreter, etc.

Ry7en, 2019-04-11:
Thank you for your reply; I should also have mentioned that I deviated from the tutorial by changing the wrap target to "pi0", instead of "pi3": 

`python <ELL-root>/tools/wrap/wrap.py --model_file model.ell --language python --target pi0`

I didn't setup miniconda as described under the section "Python 3.4 via Miniconda", or follow the steps under "OpenCV", skipping them in accordance with this guidance:

"Note: Miniconda is not available for Raspberry Pi Zero. Turns out Rasbian Stretch already comes with Python 3.5, so we only really need Miniconda to get our build of OpenCV which we use in the tutorials. On Raspberry Pi Zero you will need to jump to these instructions on how to Build OpenCV for Python 3.5"

I couldn't find 

`export PATH="/home/pi/miniconda3/bin:$PATH"`

in my .bashrc, and `which python3` returns `/usr/bin/python3` as expected.

lovettchris, 2019-04-11:
Sounds good then you only need to change this to use python3:
```
python3 <ELL-root>/tools/wrap/wrap.py --model_file model.ell --language python --target pi0
```

Ry7en, 2019-04-11:
If I interpreted the tutorial correctly, the command 

`python3 <ELL-root>/tools/wrap/wrap.py --model_file model.ell --language python --target pi0`

is intended to be run on a PC from an anaconda prompt; typing in "python3" in either `<base>` or `<py36>` mode presents me with:

`'python3' is not recognized as an internal or external command, operable program or batch file. `

lovettchris, 2019-04-29:
Ah, sorry, yes this is all quite confusing.  You have a PC that is running the ELL compiler, let's assume you set that up using Anaconda (or miniconda) and you have the recommended py36 there, so our python tools like "wrap.py" are all running fine and you can

closed 2020-12-02 by lovettchris