microsoft/ELL #213: Update documentation to use gcc-8

dashesy, 2019-04-16:
Before `cmake` is invoked [the documentation](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md) for Ubuntu16 should be updated to use gcc-8 and g++-8, e.g.

```bash
$ export CXX=`which g++-8`
$ export CC=`which gcc-8`
```

Otherwise old gcc will be used, and `make` will result in many errors

lovettchris, 2019-04-23:
Sure, or you could do this to change your system default:
```
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 50 --slave /usr/bin/g++ g++ /usr/bin/g++-8
```

dashesy, 2019-04-23:
That too works but `update-alternatives` changes gcc, and then I have to remember to change it back. Because most other projects need system gcc.

lovettchris, 2020-12-03:
Documentation updated by [this commt](https://github.com/microsoft/ELL/commit/0171dbd72a2d9643991c7f8704a0f20a77209845) to show explicit selection of gcc-8 when invoking cmake.

closed 2020-12-03 by lovettchris