microsoft/ELL #214: Setting cross-compiler for Compile command

saurabhk2, 2019-04-19:
Using ELL I was able to create a header file and obj file for my model file build in cntk. I included these files in my Azure sphere project and tried running the predictions. I see that the predictions are not coming out correct.

To create the .h and obj I have given below commands

compile -imap test_model.ell -cfn Predict -cmn Testing --bitcode -od . --fuseLinearOps True --header --blas false --optimize true --target custom --numBits 32 --cpu cortex-a7 --triple armv7--linux-gnueabihf --features +vfp4,+d16

opt.exe test_model.bc -o test_model.opt.bc -O3

llc.exe test_model.opt.bc -o test_model.obj -O3 -filetype=obj -mtriple=armv7--linux-gnueabihf -mcpu=cortex-a7 -relocation-model=pic -float-abi=hard -mattr=+vfp4,+d16

All the above commands executed fine and the .h and .obj were created. I am puzzled from where do these commands pick up the cross-compiler. To test this I removed my arm cross-compiler from the PATH variable but all the above commands still work fine and generate the .h and .obj file. How did the compile tool pick up the cross-compiler? I am suspecting that it used the default host compiler and that is the reason I am not getting the predictions right.

How do I make sure that the above commands are using the right compiler? Also is there any specific cross-compiler version that I should be using. Since the target is Azure sphere, can I point the above commands to use the cross-compiler used by Visual studio?

lovettchris, 2019-04-20:
ELL uses it's own copy of LLVM to do the cross-compiling.  On windows you will see this in this folder: ell\external\LLVMNativeWindowsLibs.x64.6.0.1 which is populated from nuget.org.  So your "compile" command line looks correct, these are the same options I used to get an azure sphere demo working.  I also assume your final azure sphere app compiled and linked correctly, so the .obj code must be fine or the linker would have complained.

So if the model is not working, I would recommend debugging this on your windows machine.  Can you get a simple windows app to work, perhaps using the ELL python module using steps similar to those outlined in our [Getting started with image classification tutorial](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/).  In this case you will use "--target host", and the "wrap.py" command lines and so on.  Also did you get any warnings from our CNTK importer?  If you can share your CNTK model I can help debug.

saurabhk2, 2019-04-25:
Thanks for the reply Chris. 
I had picked up the commands from compile.cmd in your project https://gist.github.com/lovettchris/fc4994258d92e7cd262eeace26c4f3b3
Yes I was able to compile and link correctly and the .obj and file linked with my Sphere code. I was able to get the predictions but they are not correct for all the inputs that I give. The same inputs give correct results when I try the predictions in CNTK.
I have been trying to use the ELL model in python on my laptop (used the same tutorial that you mentioned) but somehow I have been running into an error. "TypeError: Predict() missing 1 required positional argument: 'Block79'" . This is in the predict function in model.py that is generated by the tool. It seems to require a output parameter to function _model_wrapper.Predict(inputVector). I gave it a Floatvector output param. After this the error was resolved but the predict function returned null. I had not got any warnings from the CNTK importer. If you can please let me know your contact email I can email you the model file.

lovettchris, 2019-04-25:
Great sounds like some progress.  The Block79 thing is a known but, we are pushing out a fix this week to github.  My email alias at Microsoft is clovett.  Thanks.

saurabhk2, 2019-04-29:
Chris I have emailed you the model file.

lovettchris, 2019-04-29:
Saurabh, thanks for the model.  Found that with our latest version 2.5.0 this python code works on the compiled ELL model:
```python
import os
import sys
import numpy as np

script_dir = os.path.dirname(__file__)
sys.path += [os.path.join(script_dir, "host")]
sys.path += [os.path.join(script_dir, "host", "build", "release")]

import model

def softmax(x):
    e_x = np.exp(x - np.max(x))
    return e_x / e_x.sum(axis=0)

w = model.ModelWrapper()

for x in np.array([[1.16127241, -0.84578836],
                   [-0.98647612, -1.02565551],
                   [-0.62027913, -1.37240195],
                   [0.60449684,  0.37194636]]).astype(np.float32):
    inputVector = model.FloatVector(x)
    z = w.Predict(inputVector)
    y_pred = softmax(np.array(z))
    y_pred = (y_pred == y_pred.max(axis=0)).astype(int)
    print(y_pred)
```
This outputs:
```
[0 0 0 1]
[1 0 0 0]
[0 1 0 0]
[0 0 1 0]
```
I'm guessing you forgot the softmax which removes the negative values.  I compiled the ELL model with this command line:
```
python d:\git\ell\ell\tools\wrap\wrap.py --model_file cntk.ell --module_name model
```
Notice I'm being careful here to rename the module "model" because I want to do "import model" and not "import cntk" which would have been the default name given to a file named "cntk.ell".
```

saurabhk2, 2019-05-03:
Thanks Chris for your help. Doing softmax did the trick. I was not aware that we have to do softmax after the predict call. The python code also worked with ELL 2.5.0.

closed 2019-05-03 by saurabhk2

lovettchris, 2019-05-03:
Happy to help.  Thanks for closing the issue.