microsoft/ELL #215: Import error "ELL build folder not found"

dashesy, 2019-04-26:
Why is
 
```bash
make _ELL_python
```

needed as a separate build instruction? If Python 3.+ is found can we always build it?

I build with 

```bash
mkdir build; cd build
cmake ..
make
```

and it succeeds, but then I get this error

```bash
$ python tools/importers/onnx/onnx_import.py ~/model.onnx
Traceback (most recent call last):
  File "tools/importers/onnx/onnx_import.py", line 23, in <module>
    import find_ell
  File "/home/ehazar/repos/ELL/tools/importers/onnx/../../utilities/pythonlibs/find_ell.py", line 55, in <module>
    ell_build_dir = find_ell_build()
  File "/home/ehazar/repos/ELL/tools/importers/onnx/../../utilities/pythonlibs/find_ell.py", line 49, in find_ell_build
    raise Exception("ELL build folder not found in {}\nFound:{}".format(head, os.listdir(head)))
Exception: ELL build folder not found in /home/ehazar/repos/ELL
Found:['SETUP-Windows.cmd', 'build', '.travis.yml', '.gitignore', 'tools', 'Dockerfile', '.editorconfig', '.git', 'Doxyfile', 'Pipfile', 'libraries', 'rebuild.cmd', 'INSTALL-Ubuntu.md', 'CMake', 'docs', '.flake8', 'build.cmd', 'Versioning.md', 'requirements.txt', 'INSTALL-Advanced.md', 'LICENSE.txt', '.clang-format', 'interfaces', '_config.yml', 'CMakeLists.txt', 'INSTALL-Mac.md', 'StyleGuide.md', 'Gemfile', 'INSTALL-Windows.md', '.gitattributes', 'SETUP-Ubuntu.sh', 'VERSION', 'build.sh', 'README.md', 'examples', 'Pipfile.lock', 'external']
```

The error is confusing, because there is clearly a `build` directory there and is also in `Found` in the error message. So I do not know what is the error.

dashesy, 2019-04-26:
I had to install also `-dev` for Python3.6 **before** cmake. In Ubuntu16:

```bash
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt-get update
sudo apt-get install python3.6
sudo apt-get install python3.6-dev
```

closed 2019-04-26 by dashesy

lovettchris, 2019-04-26:
Interesting - are you using the miniconda setup outlined in INSTALL-Ubuntu.md ?

dashesy, 2019-04-26:
No I installed Python3.6 with apt. You are correct Miniconda would have been preferable, with no issues.

lovettchris, 2019-04-26:
ok, cool, thanks for the info.

dashesy, 2019-04-30:
Btw the reason I try avoiding conda is that it ships its own libraries, that some times could cause hard to understand issues.

This is a warning for example, that I see when using conda environment:

```text
  Cannot generate a safe runtime search path for target debugCompiler because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libffi.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/ehsan/miniconda3/lib

  Some of these libraries may not be found correctly
```