microsoft/ELL #216: Error while make

mr-yamraj, 2019-04-29:
I am using macOS 10.14.4 and after running make I am getting the below error at about [83%]:
What do I need to do?

`Scanning dependencies of target value_test
[ 83%] Building CXX object libraries/value/CMakeFiles/value_test.dir/test/src/main.cpp.o
In file included from /Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/test/src/main.cpp:11:
In file included from /Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/ComputeContext.h:12:
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/FunctionDeclaration.h:200:27: error: no matching constructor for initialization of 'detail::Function<Scalar ()>' (aka 'ell::value::detail::Function<ell::value::Scalar ()>')
        return DefineImpl(FUNCTION_TYPE{ std::forward<Fn>(fn) });
                          ^            ~~~~~~~~~~~~~~~~~~~~~~~~
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/FunctionDeclaration.h:192:23: note: expanded from macro 'FUNCTION_TYPE'
#define FUNCTION_TYPE detail::Function
                      ^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/test/src/main.cpp:183:20: note: in instantiation of function template specialization 'ell::value::FunctionDeclaration::Define<std::__1::function<ell::value::Scalar ()> &>' requested here
                  .Define(defineFunction);
                   ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/functional:1627:5: note: candidate inherited constructor not viable: no known conversion from 'std::__1::function<ell::value::Scalar ()>' to 'std::nullptr_t' (aka 'nullptr_t') for 1st argument
    function(nullptr_t) _NOEXCEPT : __f_(0) {}
    ^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/FunctionDeclaration.h:150:38: note: constructor from base class 'function<ell::value::Scalar ()>' inherited here
            using std::function<Fn>::function;
                                     ^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/FunctionDeclaration.h:148:16: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'std::__1::function<ell::value::Scalar ()>' to 'const ell::value::detail::Function<ell::value::Scalar ()>' for 1st argument
        struct Function : public std::function<Fn>
               ^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/FunctionDeclaration.h:148:16: note: candidate constructor (the implicit move constructor) not viable: no known conversion from 'std::__1::function<ell::value::Scalar ()>' to 'ell::value::detail::Function<ell::value::Scalar ()>' for 1st argument
        struct Function : public std::function<Fn>
               ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/functional:1631:5: note: candidate template ignored: requirement '__callable<function<Scalar ()> >::value' was not satisfied [with _Fp = std::__1::function<ell::value::Scalar ()>]
    function(_Fp);
    ^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/FunctionDeclaration.h:148:16: note: candidate constructor (the implicit default constructor) not viable: requires 0 arguments, but 1 was provided
        struct Function : public std::function<Fn>
               ^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/test/src/main.cpp:185:27: error: expected '(' for function-style cast or type construction
    auto rc = fn().Get<int>();
                       ~~~^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/test/src/main.cpp:185:29: error: expected expression
    auto rc = fn().Get<int>();
                            ^
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/test/src/main.cpp:202:8: error: no matching member function for call to 'Define'
    fn.Define(defineFunction);
    ~~~^~~~~~
/Users/yash/Downloads/MSR/ELL-2.5.0/libraries/value/include/FunctionDeclaration.h:198:48: note: candidate template ignored: substitution failure [with Fn = std::__1::function<ell::value::Scalar ()> &]
    [[maybe_unused]] auto FunctionDeclaration::Define(Fn&& fn)
                                               ^
4 errors generated.
make[2]: *** [libraries/value/CMakeFiles/value_test.dir/test/src/main.cpp.o] Error 1
make[1]: *** [libraries/value/CMakeFiles/value_test.dir/all] Error 2
make: *** [all] Error 2`

lovettchris, 2019-04-29:
thanks, we are investigating.

gutruaircraft, 2019-07-23:
I am using Windows 7, visual studio 2019, I got an error when I run the command "cmake -G "Visual Studio 15 2019 Win64" -Thost=x64 .."
And I got this
===================================================================
   CMake Error: Could not create named generator Visual Studio 15 2019 Win64

Generators
* Visual Studio 16 2019        = Generates Visual Studio 2019 project files.
                                 Use -A option to specify architecture.
  Visual Studio 15 2017 [arch] = Generates Visual Studio 2017 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 14 2015 [arch] = Generates Visual Studio 2015 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 12 2013 [arch] = Generates Visual Studio 2013 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 11 2012 [arch] = Generates Visual Studio 2012 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 10 2010 [arch] = Generates Visual Studio 2010 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Visual Studio 9 2008 [arch]  = Generates Visual Studio 2008 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Borland Makefiles            = Generates Borland makefiles.
  NMake Makefiles              = Generates NMake makefiles.
  NMake Makefiles JOM          = Generates JOM makefiles.
  Green Hills MULTI            = Generates Green Hills MULTI files
                                 (experimental, work-in-progress).
  MSYS Makefiles               = Generates MSYS makefiles.
  MinGW Makefiles              = Generates a make file for use with
                                 mingw32-make.
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  Watcom WMake                 = Generates Watcom WMake makefiles.
  CodeBlocks - MinGW Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles JOM
                               = Generates CodeBlocks project files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - MinGW Makefiles   = Generates CodeLite project files.
  CodeLite - NMake Makefiles   = Generates CodeLite project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Sublime Text 2 - MinGW Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - NMake Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.
  Kate - MinGW Makefiles       = Generates Kate project files.
  Kate - NMake Makefiles       = Generates Kate project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Eclipse CDT4 - NMake Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - MinGW Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.
===================================================================
and ell.sln and other files are not generated.

kernhanda, 2019-07-23:
I believe you need to run

`cmake -G "Visual Studio 16 2019" -Thost=x64 ..`

kernhanda, 2019-07-23:
You're mixing the generator name of VS 2017 and VS 2019.

lovettchris, 2019-07-23:
The command line for VS 2019 is:
```
cmake -G "Visual Studio 16 2019" -A x64 -Thost=x64
```
Make sure you run this from a VS 2019 command prompt:

[image]alias

closed 2020-12-02 by lovettchris