microsoft/ELL #217: exception: Trying to map port output to output of different size

dashesy, 2019-04-30:
After converting an ONNX model to ELL I get this when trying wrap

```bash
$ python ./tools/wrap/wrap.py -f ~/Downloads/tag.ell --target host
copy "/home/ehsan/devel/ELL/CMake/OpenBLASSetup.cmake" "host/OpenBLASSetup.cmake"
compiling model...
/home/ehsan/devel/ELL/build/bin/compile -imap /home/ehsan/Downloads/tag.ell -cfn Predict -cmn tag --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas true --optimize true
exception: Trying to map port output to output of different size, expecting 200704, but found 215296
command /home/ehsan/devel/ELL/build/bin/compile failed with error code 1
### WrapException: <class 'buildtools.EllBuildToolsRunException'>: /home/ehsan/devel/ELL/build/bin/compile -imap /home/ehsan/Downloads/tag.ell -cfn Predict -cmn tag --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas true --optimize true```

Is there a better way to see what node causes the exception?

lovettchris, 2019-05-03:
Sounds like a bug, but I have two ideas, one is to add "-v" to the wrap.py call to get verbose output which includes the "compile" command line, then run that compile command line again adding "--verbose" to get more output during compile.  You can also run "print.exe" from the ell build output folder with various options to look more closely at what is in tag.ell, and print.exe has an interesting option "--refineIterations" which applies some of the compiler refinement steps to the model generating a lower level set of compilable nodes, in case that is where the error is popping up...

lovettchris, 2019-05-11:
**Update**:  we have a potential fix in the works... it has something to do with "ReorderDataNode"...  so if you see ReorderDataNode showing up in your refined graph, then it might be the same issue.