microsoft/ELL #221: BLAS issue due to CPU

dpgindler, 2019-05-28:
When building ELL, I am getting BLAS library not found and lots of link errors.

-- Processor family: 6, model: 44
-- Unknown processor, disabling OpenBLAS
-- BLAS library not found

My CPU is an Intel Xeon 5500 Series.   This is an older CPU built code name was Gainestown or Nehalem-E.   and thus not Haswell or Skylake as desired by the cmake file.

I saw that BLAS was optional.  Wondering how I can skip building it, if that is true.

I have another option which is to try my laptop rather than desktop.   It is newer, so maybe has a supported CPU.  Input appreciated.

dpgindler, 2019-05-28:
Got past this issue by using my laptop which has a haswell based cpu.

current issue is now link errors related to python.

lovettchris, 2019-05-28:
What operating system are you running? ELL probably requires a 64 bit operating system.

dpgindler, 2019-05-28:
Windows 10 build 1809.      Seems to be minimal version for IOT edge dev.

I am thinking the issue may be that the default installer for Python is 32bit ( or at least that is what I installed ), but this solution uses 64bit...   and thus not matching up?

lovettchris, 2019-05-28:
Ah, you will need 64bit python and a 64 bit OS.  From your command prompt you should see PROCESSOR_ARCHITECTURE=AMD64.  You can probably also get open blas working on your desktop machine with older CPU if you follow our [wiki directions for building OpenBlas](https://github.com/microsoft/ELL/wiki/Building-OpenBlas-on-Windows).

closed 2020-12-02 by lovettchris