microsoft/ELL #223: Square operation returns zero vector

mr-yamraj, 2019-05-30:
Hi I am trying to debug the file make_featurizer.py in tools/utilites/pythonlibs/audio/training

and when I am using the flag --power_spec it is returning zero values in the square node.
I checked the values of last_node before this operation and I has values greater than zero. But when I am performing square using the below code it is returning zero vector.
square_node = builder.AddUnaryOperationNode(
            ell_model, ell.nodes.PortElements(last_node.GetOutputPort("output")),
            ell.nodes.UnaryOperationType.square)

mr-yamraj, 2019-05-30:
Other functions like sqrt, exp, tanh are working fine

mr-yamraj, 2019-05-30:
In the same file the output that we get from the mel filterbank does not match with the output of the mel bank that I applied in python. I used https://github.com/jameslyons/python_speech_features/blob/master/python_speech_features/base.py#L149 as refrence to calculate the fbanks and the algorithm they have used to calculate the mel filter bank is correct.

mr-yamraj, 2019-05-30:
Okay! found the bug for FFT will give a pull request but I am not finding the bug of the square (in my first comment)

kernhanda, 2019-05-31:
This should be fixed in [v2.5.4](https://github.com/microsoft/ELL/releases/tag/v2.5.4)

mr-yamraj, 2019-06-05:
Thanks! It works now.

closed 2019-06-05 by mr-yamraj