microsoft/ELL #230: ell.sln and other files are not generated

gutruaircraft, 2019-07-23:
I am using Windows 7, visual studio 2019, I got an error when I run the command "cmake -G "Visual Studio 15 2019 Win64" -Thost=x64 .."
And I got this
===================================================================
   CMake Error: Could not create named generator Visual Studio 15 2019 Win64

Generators
* Visual Studio 16 2019        = Generates Visual Studio 2019 project files.
                                 Use -A option to specify architecture.
  Visual Studio 15 2017 [arch] = Generates Visual Studio 2017 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 14 2015 [arch] = Generates Visual Studio 2015 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 12 2013 [arch] = Generates Visual Studio 2013 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 11 2012 [arch] = Generates Visual Studio 2012 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 10 2010 [arch] = Generates Visual Studio 2010 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Visual Studio 9 2008 [arch]  = Generates Visual Studio 2008 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Borland Makefiles            = Generates Borland makefiles.
  NMake Makefiles              = Generates NMake makefiles.
  NMake Makefiles JOM          = Generates JOM makefiles.
  Green Hills MULTI            = Generates Green Hills MULTI files
                                 (experimental, work-in-progress).
  MSYS Makefiles               = Generates MSYS makefiles.
  MinGW Makefiles              = Generates a make file for use with
                                 mingw32-make.
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  Watcom WMake                 = Generates Watcom WMake makefiles.
  CodeBlocks - MinGW Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles JOM
                               = Generates CodeBlocks project files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - MinGW Makefiles   = Generates CodeLite project files.
  CodeLite - NMake Makefiles   = Generates CodeLite project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Sublime Text 2 - MinGW Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - NMake Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.
  Kate - MinGW Makefiles       = Generates Kate project files.
  Kate - NMake Makefiles       = Generates Kate project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Eclipse CDT4 - NMake Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - MinGW Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.
===================================================================
and ell.sln and other files are not generated.

_Originally posted by @gutruaircraft in https://github.com/microsoft/ELL/issues/216#issuecomment-514050733_

lovettchris, 2019-07-23:
cmake was changed in this version of VS such that the "Win64" suffix was moved to a separate -A argument, so the command line for VS 2019 is :
```
cmake -G "Visual Studio 16 2019" -A x64 -Thost=x64
cmake --build . --config Release -- /m:8 /verbosity:minimal
```
Note: when you get a new version of VS you can always run "cmake -G" to learn about the new generators:
```
Generators
* Visual Studio 16 2019        = Generates Visual Studio 2019 project files.
                                 Use -A option to specify architecture.
  Visual Studio 15 2017 [arch] = Generates Visual Studio 2017 project files.
                                 Optional [arch] can be "Win64" or "ARM".
```

gutruaircraft, 2019-07-23:
Thanks for the response and I again the error specified below,
 E:\guru_clg\bit\vision_lab\embeded\ELL\build>cmake -G "Visual Studio 16 2019" -A x64 -Thost=x64
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "E:/guru_clg/bit/vision_lab/embeded/ELL/build"
 does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

gutruaircraft, 2019-07-23:
thanks for the response, I again got the error specified below,

E:\guru_clg\bit\vision_lab\embeded\ELL\build>cmake -G "Visual Studio 1
6 2019" -A x64 -Thost=x64
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "E:/guru_clg/bit/vision_lab/embeded/ELL/build"
 does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

lovettchris, 2019-07-23:
Right, you need to add ".." as well. What we normally do in ELL is create a "build" folder for the build output which means the path to the top level CMakeLists.txt file is ".." so the complete command line is:
```
cmake -G "Visual Studio 16 2019" -A x64 -Thost=x64 ..
```

lovettchris, 2019-07-23:
The latest ELL release adds support for VS 2019 so you can just type "rebuild.cmd"

lovettchris, 2019-12-29:
docs are updated to show support of VS 2019.

closed 2019-12-29 by lovettchris