microsoft/ELL #231: import _model ImportError: DLL load failed: The specified module could not be found.

ankit3600, 2019-07-30:
I am following the below tutorial:
https://microsoft.github.io/ELL/tutorials/Getting-started-with-audio-keyword-spotting-on-the-Raspberry-Pi/

(py36) E:\ELL\ELL\model>python run_classifier.py --featurizer compiled_featurizer/mfcc --classifier compiled_classifier/model --categories categories.txt --wav_file audio --sample_rate 16000 --auto_scale
Traceback (most recent call last):
File "run_classifier.py", line 118, in 
args.sample_rate, args.speaker, args.auto_scale, args.reset)
File "run_classifier.py", line 59, in test_keyword_spotter
predictor = classifier.AudioClassifier(classifier_model, categories, threshold, SMOOTHING)
File "C:\ELL\ELL\model\classifier.py", line 60, in init
self.model = ell.CompiledModel(model_path)
File "C:\ELL\ELL\model\compiled_ell_model.py", line 29, in init
self.module = importlib.import_module(model_name)
File "C:\Users\anmish\AppData\Local\Continuum\miniconda3\envs\py36\lib\importlib_init_.py", line 126, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
File "", line 994, in _gcd_import
File "", line 971, in _find_and_load
File "", line 955, in _find_and_load_unlocked
File "", line 665, in _load_unlocked
File "", line 678, in exec_module
File "", line 219, in _call_with_frames_removed
File "compiled_classifier\model.py", line 11, in 
import _model
ImportError: DLL load failed: The specified module could not be found.

I am getting the error on windows machine. I tried with/without (both) the OpenBLASLibs bin path as suggested [here](https://github.com/microsoft/ELL/issues/148). It didn't worked.
set PATH=%PATH%;C:\ELL\ELL\external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\bin

kernhanda, 2019-07-30:
Have you gone through the instructions at
https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md and built
everything, including the Python bindings? If not, please try that first.

ankit3600, 2019-07-30:
Yes Python bindings are build already. Pls find the python build log [here.](https://drive.google.com/open?id=1SQ6AiJUJcljvo3NzHtej8SoYfNztnEu8)

Although i am unable to run the python tests:

(py36) C:\ELL\ELL\build>ctest . --build-config release -R ell-python-interface-test
Test project C:/ELL/ELL/build
    Start 45: ell-python-interface-test
1/1 Test #45: ell-python-interface-test ........***Failed    0.34 sec
0% tests passed, 1 tests failed out of 1
Total Test time (real) =   0.79 sec
The following tests FAILED:
         45 - ell-python-interface-test (Failed)
Errors while running CTest

kernhanda, 2019-07-30:
Please run the following command:

ctest . -VV --build-config release -R ell-python-interface-test

ankit3600, 2019-07-30:
Still failing :(
It is opening python3.exe. Ideally it should open python.exe so that anaconda's python.exe can be opened in the directory C:\Users\anmish\AppData\Local\Continuum\miniconda3\envs\py36\python.exe

(py36) C:\ELL\ELL\build>ctest . -VV --build-config release -R ell-python-interface-test
UpdateCTestConfiguration  from :C:/ELL/ELL/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :C:/ELL/ELL/build/DartConfiguration.tcl
Test project C:/ELL/ELL/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 45
    Start 45: ell-python-interface-test

45: Test command: C:\Users\anmish\AppData\Local\Microsoft\WindowsApps\python3.exe "test.py"
45: Test timeout computed to be: 10000000
45: Python was not found but can be installed from the Microsoft Store: https://go.microsoft.com/fwlink?linkID=2082640
1/1 Test #45: ell-python-interface-test ........***Failed    0.38 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.84 sec

The following tests FAILED:
         45 - ell-python-interface-test (Failed)
Errors while running CTest

ankit3600, 2019-07-30:
Even if I fixed the path error above (changed python3.exe to python.exe in CTestTestfile.cmake file), still I got more "dll load failed" errors.

(py36) C:\ELL\ELL\build>ctest . -VV --build-config release -R ell-python-interface-test
UpdateCTestConfiguration  from :C:/ELL/ELL/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :C:/ELL/ELL/build/DartConfiguration.tcl
Test project C:/ELL/ELL/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 45
    Start 45: ell-python-interface-test

45: Test command: C:\Users\anmish\AppData\Local\Continuum\miniconda3\envs\py36\python.exe "test.py"
45: Test timeout computed to be: 10000000
45: Traceback (most recent call last):
45:   File "test.py", line 45, in <module>
45:     raise err
45:   File "test.py", line 10, in <module>
45:     import ell
45:   File "C:\ELL\ELL\build\interfaces\python\package\ell\__init__.py", line 47, in <module>
45:     from . import data
45:   File "C:\ELL\ELL\build\interfaces\python\package\ell\data\__init__.py", line 9, in <module>
45:     from ..ell_py import AutoDataVector, \
45:   File "C:\ELL\ELL\build\interfaces\python\package\ell\ell_py.py", line 11, in <module>
45:     import _ell_py
45: ImportError: DLL load failed: %1 is not a valid Win32 application.
1/1 Test #45: ell-python-interface-test ........***Failed    1.47 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   1.56 sec

The following tests FAILED:
         45 - ell-python-interface-test (Failed)
Errors while running CTest

ankit3600, 2019-08-01:
The problem is solved by installing miniconda3-4.5.4-Windows-x86_64.exe from here https://repo.continuum.io/miniconda/

It was happening because latest miniconda uses python3.7 but ELL requires python 3.6. So downgrade the miniconda version

closed 2019-08-01 by ankit3600