microsoft/ELL #232: Issues with "Training an audio keyword spotter with PyTorch" tutorial

psiphi75, 2019-08-07:
I have followed the *Training an audio keyword spotter with PyTorch* tutorial, on a clean install of Ubuntu 18.04.  

I have found the following minor issues:

- The tutorial did not mention to install `pyaudio`.
- The tutorial did not mention to install `onnx` - I used the following command `conda install -c conda-forge onnx=1.4.1`
- The tutorial has a few typos and other mistakes, but I could easily navigate around them (e.g. "GRUKeywordSpotter.ell" in the tutorial is actually "GRU128KeywordSpotter.ell")

lovettchris, 2019-08-07:
Good point.  The pyaudio setup was mentioned in the [first audio tutorial](https://microsoft.github.io/ELL/tutorials/Getting-started-with-audio-keyword-spotting-on-the-Raspberry-Pi/).  ONNX setup is missing from all our docs.  Thanks for reporting these doc bugs.

lovettchris, 2020-12-30:
Most of these issues are fixed and "onnx" is listed in the requirements.txt file in the ELL repo.

closed 2020-12-30 by lovettchris