microsoft/ELL #233: `make_dataset.py` does not accept --auto-scale and --sample-rate in Linux

psiphi75, 2019-08-07:
The "Training an audio keyword spotter with PyTorch" mentions to use the `make_dataset.py` script, as follows:

```sh
python make_dataset.py --list_file audio/training_list.txt --featurizer compiled_featurizer/mfcc --sample_rate 16000 --window_size 40 --shift 40 --auto_scale
```

However, when running this on Ubuntu 18.04, it errors with the following message:

```text
usage: make_dataset.py [-h] [--list_file LIST_FILE] [--outdir OUTDIR]
                       [--categories CATEGORIES] [--featurizer FEATURIZER]
                       [--window_size WINDOW_SIZE] [--shift SHIFT]
                       [--noise_path NOISE_PATH]
                       [--max_noise_ratio MAX_NOISE_RATIO]
                       [--noise_selection NOISE_SELECTION]
make_dataset.py: error: unrecognized arguments: --auto_scale --sample_rate 16000
```

Hence `--auto_scale` and `--sample_rate` are invalid parameters.

lovettchris, 2019-08-07:
Yes, apologies the docs are out of sync.  The auto_scale and sample_rate properties are now stored inside the featurizer.ell file created by make_featurizer.py, so no need to provide that in make_dataset.py any more.  If you have an old featurizer model that is missing this metadata then you may need to rerun make_featurizer.py, we'll update the docs.

psiphi75, 2019-08-07:
Cool, I'll close this case, since it's covered by #232.

Thanks.

closed 2019-08-07 by psiphi75