microsoft/ELL #239: Compile pretained model failed in C++ tutorial

kunhung, 2019-11-19:
Hi,

while following the steps from C++ tutorial, failed to compile the pretrained model "model.ell" with below error, please help. Thanks!

c:\test\ELL-master\build\tools\wrap>python wrap.py --model_file model.ell --language cpp --target host --outdir model
copy "c:\test\ELL-master\CMake/OpenBLASSetup.cmake" "model\OpenBLASSetup.cmake"
compiling model...
C:/test/ELL-master/build/bin/release/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od model --fuseLinearOps True --optimizeReorderDataNodes True --header --blas true --optimize true --parallelize true --vectorize
Command line parse error:
Cannot read from specified input map file: model.ell
command C:/test/ELL-master/build/bin/release/compile failed with error code 1
### WrapException: <class 'buildtools.EllBuildToolsRunException'>: C:/test/ELL-master/build/bin/release/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od model --fuseLinearOps True --optimizeReorderDataNodes True --header --blas true --optimize true --parallelize true --vectorize

kunhung, 2019-11-19:
C:\test\ELL-master\build\bin\Release>compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas false --optimize true
Command line parse error:
Cannot read from specified input map file: model.ell

kunhung, 2019-11-19:
issue resolved

closed 2019-11-19 by kunhung

shalaka188, 2021-03-13:
i also am facing the issue realted to this compiling using wrap . It is access denied  error will u plz help to solve this error
thank you