microsoft/ELL #241: Is ELL applied to Azsure Sphere MCU?

kunhung, 2019-11-19:
Hi,

we are trying to integrate Azsure Sphere MCU as IoT edge and wondering if ELL can be applied to Sphere MCU as edge side AI analysis? if yes, is there any instruction or sample? Thanks!

Ken

lovettchris, 2019-11-19:
The attached sample runs a keyword spotter on the Azure Sphere mt3620 ARM Cortex-M4 processor, on a statically linked audio sample (defined in seven.h).

[m4_keywordspotter.zip](https://github.com/microsoft/ELL/files/3865672/m4_keywordspotter.zip)

kunhung, 2019-11-25:
Hi,

Thanks for the information and the sample can be run without issue. one more question, is ELL only to be run on Real time Cortex-M4, not for Coretex-A7 in high level app? thanks!

clovett, 2019-12-29:
ELL works on Cortex-A7.

closed 2020-05-13 by lovettchris