microsoft/ELL #242: Windows setup incomplete

fleischkarotte, 2019-12-18:
Hi 
I tried to follow the guide "Building a 64-bit version of the Embedded Learning Library (ELL) on Windows" when i noticed that when running the python interface test with the command:
ctest . --build-config release -R ell-python-interface-test
it will produce an error. This is because the package numpy was not installed with Miniconda3. 
Maybe you could add this to the tutorial as I needed some considerable time to figure this out.

Numpy has to be installed outside the virtual environment py36 in order for everything to work as described in the tutorial.

Also it seems to be important which miniconda version is being used as the repurposing script is not working with python 3.7. For me it worked with Miniconda3-4.5.4

Lastly thanks for the great work!

lovettchris, 2019-12-20:
Thanks for playing with ELL and thanks for reporting your problems with miniconda.  I just tried the steps outlined in INSTALL-windows.md using the latest version of miniconda, and I also ran into problems with cmake not finding python:
```
-- Looking for python 3.7: PYTHONINTERP_FOUND=FALSE, PYTHONLIBS_FOUND=FALSE
-- Looking for python 3.6: PYTHONINTERP_FOUND=FALSE, PYTHONLIBS_FOUND=FALSE
```
This is odd because the miniconda prompt can find python no problem.  So this would indicate there is some incompatibility with the latest miniconda3 and the VS 2017 python cmake modules `c:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\share\cmake-3.12\Modules\FindPythonInterp.cmake`

Thanks for finding a version of miniconda that works properly... we should add that to the docs.  We should also move the docs to VS 2019, are you using VS 2017 or 2019 ?

fleischkarotte, 2019-12-20:
I am glad I could help you and also I have to thank you for the great tutorial! For me it was the best tutorial for setting up a neural net on the raspberry so far and it really helped me with my project!

So for me the newest version of miniconda3 worked to the point that I tried to repurpose a net from ELL gallery. I even deployed two nets on the raspberry following the steps in the tutorial using the newest version of miniconda3 but when I tried to repurpose the net there was an error message. I don't recall the error exactly but it was something with the python2.7.dll so in order to use python 2.6 I reinstalled an older version of miniconda.

Also I am using VS2017 as on the first try I didn't realise that I could also use VS2019 by simply invoking CMake with an slightly altered command.

clovett, 2019-12-29:
[this commit](https://github.com/microsoft/ELL/commit/aa491f053692f985c79335d2f3f95de754ff1349) fixes the docs to point to VS 2019 which works properly with latest miniconda.

fleischkarotte, 2020-01-06:
Thanks for fixing the tutorial!

closed 2020-01-06 by fleischkarotte