microsoft/ELL #244: Error Compiling ELL

victorbutuza, 2020-02-27:
**C:\ELL\libraries\nodes\include\ConstantNode.h(271,123): error C2794: 'PortMemoryLayout': is not a member of any direct or indirect base class of '<Unknown>' (compiling source file C:\ELL\libraries\nodes\src\WinogradConvolutionNode.cpp) [C:\ELL\build\libraries\nodes\nodes.vcxproj]**

C:\ELL\libraries\nodes\src\WinogradConvolutionNode.cpp(1154): 
message : while compiling class template member function 'bool ell::nodes::WinogradConvolutionNode<float>::Refine(ell::model::ModelTransformer &) const' [C:\ELL\build\libraries\nodes\nodes.vcxproj]
C:\ELL\libraries\nodes\src\WinogradConvolutionNode.cpp(1530): message : see reference to class template instantiation 'ell::nodes::WinogradConvolutionNode<float>' being compiled [C:\ELL\build\libraries\nodes\nodes.vcxproj]

**C:\ELL\libraries\nodes\include\ConstantNode.h(263,123): error C2794: 'MemoryShape': is not a member of any direct or indirect base class of
 '<Unknown>' (compiling source file C:\\ELL\libraries\nodes\src\WinogradConvolutionNode.cpp) [C:\ELL\build\libraries\nodes\nodes.vcxproj]**

lovettchris, 2020-02-27:
Indeed, something must have changed in the VC++ compiler, I'll push a fix.

lovettchris, 2020-05-13:
Fix is pushed.

closed 2020-05-13 by lovettchris