microsoft/ELL #245: OpenBLAS cannot be linked

victorbutuza, 2020-02-28:
After following the wiki to compile BLAS (successfully) I still see

-- Processor family: 6, model: 94
-- Unknown processor, disabling OpenBLAS
-- BLAS library not found

which blocks the rebuild due to linking issue value.lib(VectorOperations.obj) : error **LNK2019**: unresolved external symbol "float __cdecl ell::math::Blas::Dot(int,float const *,int,float const

lovettchris, 2020-02-28:
Ah, yes, our OpenBlas support is not very comprehensive I'm afraid, so you might need to edit `ELL\CMake\OpenBLASSetup.cmake` and add a mapping for that chip, I believe that is a [Skylake ](https://en.wikichip.org/wiki/intel/cpuid#Family_6) chip which we have mapped to `sandybridge`  Can you try that and see if it works?  Just add this on line 162:
```
                    elseif(processor_model EQUAL 42 OR processor_model EQUAL 92)
```

victorbutuza, 2020-02-28:
Works fine thanks Just one mention  processor_model EQUAL 94
Compiled successfully.

closed 2020-02-28 by victorbutuza