microsoft/ELL #247: cMake is not recognized as an internal or external command

MSD077, 2020-05-31:
Hello
       I'm working on ELL and have installed visual studio 2019. But when i try to invoke cMake through the following command

cmake -G "Visual Studio 16 2019" -A x64 -T host=x64 -D ONNX=ON ..

I get an error that says cMake is not recognized as an internal or external command. 

My cMake exe file is in the following location: C:\Program Files (x86)\Microsoft Visual Studio\2019\Preview\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe

Whereas the location where ELL repository is located is: D:\PROJECTS\Git\cmd\ELL

Can someone kindly help me out. Thanks!

byronChanguion, 2020-06-01:
Looks like you don't have cmake in your path.

2 suggestions:
1. Run the command from a Visual Studio command prompt e.g. Start -> Visual Studio 2019 -> "Developer Command Prompt for VS2019". The environment for this command prompt is now properly configured for Visual Studio build tools and CMake.
Or,
2. Activate the Conda environment you created for use with ELL (e.g. "activate py36"). Use Conda to install CMake (e.g. "conda install -c anaconda cmake"). Now you can run CMake anytime after activating your Conda environment.

MSD077, 2020-06-02:
That worked. Thanks a lot!

closed 2020-06-02 by MSD077