microsoft/ELL #248: import _model Import error

MSD077, 2020-06-02:


MSD077, 2020-06-02:
Hello,
when executing the call_model.py script, I get the following error

(py36) D:\PROJECTS\Practice Model\host>python call_model.py
Traceback (most recent call last):
  File "call_model.py", line 19, in <module>
    import model
  File "D:\PROJECTS\Practice Model\host\model.py", line 11, in <module>
    import _model
ImportError: DLL load failed: The specified module could not be found.

It will be much helpful if someone could guide me on this. Also what is the correct directory for the call_model.py script? Should it be inside the host/build folder? 

Thanks!

lovettchris, 2020-12-02:
you just need move `call_model.py` up one level in your folder hierarchy.  `call_model.py` and `tutorial_helpers.py` should live in the same folder that contains "host".  You can see what this is when you look at `tutorial_helpers.py` which is setting up a module search path for the "host" folder.

A-KL, 2020-12-02:
Hi guys, I've tried all possible location for `call_model.py` but the error is still the same:
```
Traceback (most recent call last):
  File "c:/Users/.../Sources/Python/object-tracking/ell/call_model.py", line 19, in <module>
    import model
  File "c:\Users\...\Sources\Python\object-tracking\ell\host\model.py", line 11, in <module>
    import _model
ImportError: DLL load failed while importing _model: The specified module could not be found.
```
Looks like it's unable to load one of the binaries: `_model.pyd` or `_model.lib`. I'm running on Python 3.8 & Windows 10 x64. 

Would appreciate any help. Thanks!

lovettchris, 2020-12-03:
Another possibility is missing open_blas DLL dependency.
See: Path Environment in  ELL/INSTALL-Windows.md at master · microsoft/ELL (github.com)<https://github.com/microsoft/ELL/blob/master/INSTALL-Windows.md#path-environment>