microsoft/ELL #252: Training an audio keyword spotter with PyTorch ELL

tasiddiq94, 2020-06-10:
I am following the tutorial in the following link to train a Kewyword spotter with Pytorch using ELL:
https://microsoft.github.io/ELL/tutorials/Training-audio-keyword-spotter-with-pytorch/

After training my model and saving it to the following onnx file: GRU128KeywordSpotter.onnx, I need to import it from ONNX format into the ELL format using the following command:

[Windows] python %ELL_ROOT%\tools\importers\onnx\onnx_import.py GRU128KeywordSpotter.onnx

When I run the above command, I get the following error:

MainThread [2020-06-10 07:16:37,954] Pre-processing...
MainThread [2020-06-10 07:16:37,954] loading the ONNX model from: GRU128KeywordSpotter.onnx
MainThread [2020-06-10 07:16:37,958] Loaded ONNX model in 0.002 seconds.
MainThread [2020-06-10 07:16:37,958] ONNX IR_version 6
MainThread [2020-06-10 07:16:37,959] ONNX Graph producer: pytorch version 1.5
MainThread [2020-06-10 07:16:37,959] ONNX Graph total len: 1
MainThread [2020-06-10 07:16:37,960] Input input.1 Inputs [] [] Outputs: ['input.1'] [((1, 1, 80), 'channel_row_column')] Attributes: {}
MainThread [2020-06-10 07:16:37,960] Shape Shape_0 Inputs ['input.1'] [((1, 1, 80), 'channel_row_column')] Outputs: ['11'] [((1, 1, 80), 'channel_row_column')] Attributes: {}
MainThread [2020-06-10 07:16:37,961] Constant Constant_1 Inputs [] [] Outputs: ['12'] [((1,), 'channel')] Attributes: {'tensor': '...'}
MainThread [2020-06-10 07:16:37,961] Constant 11 Inputs [] [] Outputs: ['11'] [((3,), 'channel')] Attributes: {'tensor': '...'}
MainThread [2020-06-10 07:16:37,961] Gather Gather_2 Inputs ['11', '12'] [((1, 1, 80), 'channel_row_column'), ((1,), 'channel')] Outputs: ['13'] [((1, 80), 'row_column')] Attributes: {'axis': 0}
MainThread [2020-06-10 07:16:37,962] Unsqueeze Unsqueeze_3 Inputs ['13'] [((1, 80), 'row_column')] Outputs: ['17'] [((1, 1, 80), 'channel_row_column')] Attributes: {'axes': [0]}
Traceback (most recent call last):
  File "C:\Users\taima\ELL\tools\importers\onnx\onnx_import.py", line 95, in <module>
    main()
  File "C:\Users\taima\ELL\tools\importers\onnx\onnx_import.py", line 91, in main
    convert(args.input, args.output_directory, args.zip_ell_model, args.step_interval, args.lag_threshold)
  File "C:\Users\taima\ELL\tools\importers\onnx\onnx_import.py", line 42, in convert
    lag_threshold_msec=lag_threshold)
  File "C:\Users\taima\ELL\tools\importers\onnx\onnx_to_ell.py", line 32, in convert_onnx_to_ell
    importer_model = converter.load_model(path)
  File "C:\Users\taima\ELL\tools\importers\onnx\lib\onnx_converters.py", line 2004, in load_model
    return self.set_graph(graph)
  File "C:\Users\taima\ELL\tools\importers\onnx\lib\onnx_converters.py", line 2026, in set_graph
    node = self.get_converter(onnx_node).convert(onnx_node)
  File "C:\Users\taima\ELL\tools\importers\onnx\lib\onnx_converters.py", line 1321, in convert
    node = super().convert(node)
  File "C:\Users\taima\ELL\tools\importers\onnx\lib\onnx_converters.py", line 118, in convert
    node.output_shapes = self.get_output_shapes()
  File "C:\Users\taima\ELL\tools\importers\onnx\lib\onnx_converters.py", line 1353, in get_output_shapes
    input_shapes += [n.output_shapes[0]]
AttributeError: 'NoneType' object has no attribute 'output_shapes'

Can anybody explain what the problem is and help resolve the issue? TIA

lovettchris, 2020-07-28:
Thanks for the report, I see that there have been significant changes to ONNX since the tutorial was written, and the ELL ONNX importer needs to be updated accordingly.

lovettchris, 2020-12-03:
This is fixed by [this commt](https://github.com/microsoft/ELL/commit/0171dbd72a2d9643991c7f8704a0f20a77209845).

closed 2020-12-03 by lovettchris