microsoft/ELL #254: 44 - ell-python-interface-test (Failed)

neetha-jyothish, 2020-09-22:
I'm trying to build ELL on windows 10. However, on running the following command:

`ctest . --build-config release -R ell-python-interface-test`

I get this error:
Test project C:/ELL/ELL/build
    Start 44: ell-python-interface-test
1/1 Test #44: ell-python-interface-test ........***Failed    0.12 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.25 sec

The following tests FAILED:
         44 - ell-python-interface-test (Failed)
Errors while running CTest

lovettchris, 2020-12-02:
If you run `ctest . --build-config release -R ell-python-interface-test -VV` you will get more details on why it fails.  It could fail if you have not setup your python environment correctly, or you have some missing python packages -- see the "[requirements.txt](https://github.com/microsoft/ELL/blob/master/requirements.txt)" file for a full list.

A-KL, 2020-12-05:
Hi guys, I'm also trying to tun python tests for ELL on windows  + Python 3.8. Same error.

```
C:\Sources\ELL\build>ctest . --build-config release -R ell-python-interface-test -VV
UpdateCTestConfiguration  from :C:/Sources/ELL/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :C:/Sources/ELL/build/DartConfiguration.tcl
Test project C:/Sources/ELL/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 44
    Start 44: ell-python-interface-test

44: Test command: C:\Users\Anatolii\AppData\Local\Programs\Python\Python38\python.exe "test.py"
44: Test timeout computed to be: 10000000
44: Traceback (most recent call last):
44:   File "test.py", line 45, in <module>
44:     raise err
44:   File "test.py", line 10, in <module>
44:     import ell
44:   File "C:\Sources\ELL\build\interfaces\python\package\ell\__init__.py", line 47, in <module>
44:     from . import data
44:   File "C:\Sources\ELL\build\interfaces\python\package\ell\data\__init__.py", line 9, in <module>
44:     from ..ell_py import AutoDataVector, \
44:   File "C:\Sources\ELL\build\interfaces\python\package\ell\ell_py.py", line 11, in <module>
44:     import _ell_py
44: ImportError: DLL load failed while importing _ell_py: The specified module could not be found.
1/1 Test #44: ell-python-interface-test ........***Failed    0.08 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.20 sec

The following tests FAILED:
         44 - ell-python-interface-test (Failed)
Errors while running CTest
```

The content of ell package looks like:
```
 Directory of C:\Sources\ELL\interfaces\python\package\ell

12/02/2020  09:26 PM    <DIR>          .
12/02/2020  09:26 PM    <DIR>          ..
12/02/2020  09:26 PM               713 CMakeLists.txt
12/02/2020  09:26 PM    <DIR>          data
12/02/2020  09:26 PM    <DIR>          math
12/02/2020  09:26 PM    <DIR>          model
12/02/2020  09:26 PM    <DIR>          neural
12/02/2020  09:26 PM    <DIR>          nodes
12/02/2020  09:26 PM               293 platform.py
12/02/2020  09:26 PM             7,391 rpi_magic.py
12/02/2020  09:26 PM    <DIR>          trainers
12/02/2020  09:26 PM    <DIR>          util
12/02/2020  09:26 PM    <DIR>          vision
12/02/2020  09:26 PM             2,071 __init__.py

lovettchris, 2020-12-06:
Thanks, so the compiled ELL module named "_ell_py" is not loading.  This can happen if this path is not in your PATH environment:

```
C:\Sources\ELL\external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\bin
```

It could require this path or the sandybridge path depending on your CPU type.  The actual path needed is output when you run "cmake" to build ELL.

A-KL, 2020-12-14:
Thanks, @lovettchris. I've checked and verified my path set up:

```
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>where libopenblas.dll
C:\Sources\ELL\external\OpenBLASLibs.0.2.19.3\build\native\x64\sandybridge\bin\libopenblas.dll
```
also 

```
 Directory of C:\Sources\ELL\build\interfaces\python\Release

12/05/2020  02:50 PM    <DIR>          .
12/05/2020  02:50 PM    <DIR>          ..
12/05/2020  02:50 PM               750 _ell_py.exp
12/05/2020  02:50 PM             1,732 _ell_py.lib
12/05/2020  02:50 PM        34,397,184 _ell_py.pyd
```

However, during these tests - I didn't use Miniconda and Python 3.6. Just 3.8.5. Is it essential to run under 3.6?

n-gineer, 2020-12-19:
> Thanks, so the compiled ELL module named "_ell_py" is not loading. This can happen if this path is not in your PATH environment:
>
> ``` [...]

You mean this command, right?
`cmake --build . --config Release`
not this one?
`cmake --build . --target _ELL_python --config Release`

I don't have a folder "C:\Sources"; should I have made this directory and cloned there? I didn't see anything to this effect in the ReadMe. I'm not seeing any path mentioning OpenBLASLibs when building ELL or ELL_python. 

The haswell and sandy bridge seem to be rather old CPU's. Mine is an Intel  i7-8650U, which seems to be Kaby Lake; does this mean it isn't supported by ELL?

lovettchris, 2020-12-20:
**A-KL**, yes please try python 3.6, I just tried python 3.8 and it didn't work... so we might have a bug there to investigate.

**n-gineer,** when you run cmake it prints out something like `-- Using BLAS include library ...` take that path replace the lib at the end with "bin" and add that to your PATH environment, so that ELL can find the openblas.dll that lives there.

If you are using AMD chip, it will not use openblas since we have not shipped a version of openblas that works on AMD chips.  In that case you would have to build openblas yourself if you want that.  You can use ELL without openblas of course. 

It also prints out a bunch of `-- PYTHON_*` information showing you what version of python it found, this is the version of python that _ELL_Python will target.

You can also download this [Dependency Walker tool][https://dependencywalker.com/) and check the _ell_py.pyd to see what it thinks is missing.  I had the best luck running "depends.exe" then using file open to load the _ell_py.pyd.  PS: the depends.exe UI is very slow on this binary for some reason, so just be patient.  I find if I use View/collapseAll then the UI works better.  You can safely ignore all the missing modules that start with the name API-MS-WIN-CORE...  You'll be looking for missing dependencies in the main view, mine is ok, and looks like this.  Note, I'm using AMD so you can see there is no openblas listed.  But if you see a warning or error icon in this view then that's your missing dependency.

[image]

lovettchris, 2020-12-20:
n-gineer, also check this file in the ELL repo: "cmake\OpenBLASSetup.cmake" as you may need to edit that file to add the processor_model of your new chip like we did with model numbers 60, 63, 69, 70, 79, 78, 142, 158 binding them all toe the haswell version of openblas with "set(processor_generation "haswell")".

You can get your CPU family and model number by running this:
```
reg query HKLM\Hardware\Description\System\CentralProcessor\0 /v Identifier
```

lovettchris, 2020-12-20:
**A-KL** - my python 3.8 problem was this issue on the [latest version of numpy](https://developercommunity.visualstudio.com/content/problem/1207405/fmod-after-an-update-to-windows-2004-is-causing-a.html).  The solution is:

```
pip uninstall numpy
pip install numpy==1.19.3
```

Now my build of ELL is working fine on python 3.8.6.

n-gineer, 2020-12-22:
> **A-KL**, yes please try python 3.6, I just tried python 3.8 and it didn't work... so we might have a bug there to investigate.
>
> **n-gineer,** when you run cmake it prints out something like `-- [...]
It seems since I already built, it doesn't output the lines I'm looking for. Should I delete the build folder contents and rebuild? 

I did found this file in the ELL build directory: ell_build_tools.json, within which it gave:
`"blas": "C:/.../Autocharger/Computer Vision/ELL/external/OpenBLASLibs.0.2.19.3/build/native/x64/haswell/lib/libopenblas.dll.a"`
I had already added the haswell/bin version of this to my Path and verified it is being seen by Powershell with an echo $ENV:PATH

Here is the entirety of the output from the build command:
`andre@BOOK-2-1: build> cmake --build . --config Release                                                                                               
Microsoft (R) Build Engine version 16.8.2+25e4d540b for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  utilities.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\lib
  raries\utilities\Release\utilities.lib
  math.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\librarie
  s\math\Release\math.lib
  data.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\librarie
  s\data\Release\data.lib
  dsp.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\libraries
  \dsp\Release\dsp.lib
  emitters.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\libr
  aries\emitters\Release\emitters.lib
  value.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\librari
  es\value\Release\value.lib
  emittable_functions.vcxproj -> C:\...\Autocharger\Computer Vision\ELL
  \build\libraries\emittable_functions\Release\emittable_functions.lib
  evaluators.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\li
  braries\evaluators\Release\evaluators.lib
  functions.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\lib
  raries\functions\Release\functions.lib
  model.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\librari
  es\model\Release\model.lib
  predictors.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\li
  braries\predictors\Release\predictors.lib
  nodes.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\librari
  es\nodes\Release\nodes.lib
  trainers.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\libr
  aries\trainers\Release\trainers.lib
  common.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\librar
  ies\common\Release\common.lib
  makeExamples.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\
  bin\Release\makeExamples.exe
  Generating example models
  optimization.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\
  libraries\optimization\Release\optimization.lib
  passes.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\librar
  ies\passes\Release\passes.lib
  _ELL_python.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\i
  nterfaces\python\Release\_ell_py.pyd
  compile.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\bin\R
  elease\compile.exe
  pythonPlugins.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build
  \tools\utilities\pythonPlugins\Release\pythonPlugins.lib
  profile.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\bin\R
  elease\profile.exe
  apply.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\bin\Rel
  ease\apply.exe
  testing.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\libra
  ries\testing\Release\testing.lib
  common_test.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\l
  ibraries\common\Release\common_test.exe
  data_test.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\lib
  raries\data\Release\data_test.exe
  debugCompiler.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build
  \bin\Release\debugCompiler.exe
  model_testing.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build
  \libraries\model_testing\Release\model_testing.lib
  dsp_nodes_test.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\buil
[... 47 lines cut ...]
  protoNNTrainer.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\buil
  d\bin\Release\protoNNTrainer.exe
  retargetTrainer.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\bui
  ld\bin\Release\retargetTrainer.exe
  sweepingSGDTrainer.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\
  build\bin\Release\sweepingSGDTrainer.exe
  trainers_test.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build
  \libraries\trainers\Release\trainers_test.exe
  utilities_test.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\buil
  d\libraries\utilities\Release\utilities_test.exe
  value_test.vcxproj -> C:\...\Autocharger\Computer Vision\ELL\build\li
  braries\value\Release\value_test.exe`

Here is what dependency walker shows:

(image to be added shortly, having trouble)

There are a number of API-WIN-MS-CRT and some others hidden in the tree. The LIBOPENBLAS tree has no warnings or errors showing when I expanded fully.

> ```
> reg query HKLM\Hardware\Description\System\CentralProcessor\0 /v Identifier
> ```
returns 
`    Identifier    REG_SZ    Intel64 Family 6 Model 142 Stepping 10`
which appears to be already supported in "cmake\OpenBLASSetup.cmake"

lovettchris, 2020-12-30:
Ok, great, so your openblas setup looks great, the cmake setup is recommending haswell version of openblas and depends.exe is finding it, so now it is a complete mystery why the compiled ELL model is not loading properly in your python 3.8 environment.  

n-gineer, since you are not the one that opened this bug, can you remind me what is the exact issue you are facing?  Is it still that the "ell-python-interface-test" unit test is failing for you, namely you cannot get your python environment to load _ell_py module?  If so can you try this:

```
cd into your ELL\build\interfaces\python\package folder
python
import ELL
```

This is how you can test the ell package loads into your python environment independent of the rest of the unit test.

MoonSunBeen, 2020-12-31:
@lovettchris 
Hello, I have the same problem. 

When I enter the cmd `ctest . --build-config release -R ell-python-interface-test -VV`
It is part of my error code 
`44: DoubleVector test ... Passed
44: FloatVector test ... Passed
44: vector_test passed
44: model_test failed
44: exception:
44: Loading file ../../../examples/models/model_1.model
44: common_test failed
44: trainers_test.test -- TBD
44: trainers_test passed
44: test_tensor ... Passed
44: test_predictor ... Passed`

model_test and common_test are failed

So, I've check the requirements.txt
`(py36) C:\Program Files\ELL\interfaces\python\package\ell>python -m pip freeze
appdirs==1.4.4
audioread==2.1.9
bcrypt==3.2.0
certifi==2020.12.5
cffi==1.14.4
chardet==4.0.0
cntk==2.7
cntk-gpu==2.7
configparser==5.0.1
cryptography==3.3.1
cycler==0.10.0
dask==2020.12.0
dataclasses==0.7
decorator==4.4.2
flake8==3.8.4
future==0.18.2
idna==2.10
importlib-metadata==3.3.0
iso8601==0.1.13
joblib==1.0.0
kiwisolver==1.3.1
librosa==0.8.0
llvmlite==0.35.0
matplotlib==3.3.3
mccabe==0.6.1
numba==0.52.0
numpy==1.19.3
olefile==0.46
onnx==1.8.0
opencv-python==4.4.0.46
packaging==20.8
paramiko==2.7.2
parse==1.18.0
Pillow @ file:///C:/ci/pillow_1603822370986/work
pooch==1.3.0
protobuf==3.4.1
psutil==5.8.0
PyAudio==0.2.11
pycodestyle==2.6.0
pycparser==2.20
pyflakes==2.2.0
PyNaCl==1.4.0
pyparsing==2.4.7
python-dateutil==2.8.1
python-speech-features==0.6
PyYAML==5.3.1
requests==2.25.1
resampy==0.2.2
scikit-learn==0.24.0
scipy==1.5.4
serial==0.0.97
six @ file:///home/conda/feedstock_root/build_artifacts/six_1590081179328/work
SoundFile==0.10.3.post1
threadpoolctl==2.1.0
torch==1.7.0
torchvision==0.8.1
typing-extensions @ file:///tmp/build/80754af9/typing_extensions_1598376058250/work
urllib3==1.26.2
wincertstore==0.2
zipp==3.4.0`

I've checked all the package are installed

I have no idea about this problem. Can you suggest me the solution of this problem?

lovettchris, 2020-12-31:
MoonSunBeen, this looks like a different issue, can you open a new issue for it? Thanks.

n-gineer, 2021-01-01:
> n-gineer, since you are not the one that opened this bug, can you remind me what is the exact issue you are facing? Is it still that the "ell-python-interface-test" unit test is failing for you, [...]

I'm having the same error as OP after attempting this step of the install instructions:

`ctest . --build-config release -R ell-python-interface-test`

And it is failing on the same test, "ell-python-interface-test".

I attempted to import ELL and it returned:

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'ELL'`

lovettchris, 2021-01-01:
n-gineer, thanks can you run this as this is the lowest level test you can do to see if you can import the ELL module.

From your ELL repo:
cd build\interfaces\python\package
dir ell\Release

You should see these files:

12/29/2020  12:43 AM    <DIR>          .
12/29/2020  12:43 AM    <DIR>          ..
12/29/2020  12:43 AM               751 _ell_py.exp
12/29/2020  12:43 AM             1,732 _ell_py.lib
12/29/2020  12:43 AM        34,452,992 _ell_py.pyd

Now run "python" interactively in the same conda environment you used to build ELL:

python

then run this python statement

>>> import ell

And let me know what you see?  If this works then try this python command:

>>> dir(ell)

and you should see this:

>>> import ell
>>> dir(ell)
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__this_file_directory', '__warningregistry__', 'data', 'ell_py', 'ensure_openblas', 'init_magics', 'math', 'model', 'neural', 'nodes', 'package_dir', 'platform', 'rpi_magic', 'saved', 'trainers', 'util']

n-gineer, 2021-02-06:
Hi @lovettchris, thanks for the help on this. I'm sorry I did a poor job of reading instructions and thought the use of conda was optional, but upon re-reading I realized that python as a whole was optional. 

results of dir:
```
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---l          12/21/20   1:16 PM            962 _ell_py.exp
-a---l          12/21/20   1:16 PM           1732 _ell_py.lib
-a---l          12/21/20   1:16 PM       34391552 _ell_py.pyd
```
I had installed ell without using a python environment, so I suspect that may be causing my problems. I went back and installed miniconda and made a python environment and tried import ell there (not expecting it to work due to backwards order of operations) and got back:

```
  File "<stdin>", line 1, in <module>                                                                                     File "C:\Users\...\ELL\build\interfaces\python\package\ell\__init__.py", line 47, in <module>                                             from . import data                                                                                                    File "C:\Users\...\ELL\build\interfaces\python\package\ell\data\__init__.py", line 9, in <module>                                         from ..ell_py import AutoDataVector, \                                                                                File "C:\Users\...\ELL\build\interfaces\python\package\ell\ell_py.py", line 11, in <module>                                               import _ell_py                                                                                                      ImportError: Module use of python38.dll conflicts with this version of Python.   
```
When I tried doing it from conda prompt <base>, it worked okay (probably because I didn't properly set up the environment beforehand, and I got back:
```
>>> import ell
>>> dir (ell)
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__this_file_directory', 'data', 'ell_py', 'ensure_openblas', 'math', 'model', 'neural', 'nodes', 'package_dir', 'platform', 'saved', 'trainers', 'util']
```

I'm guessing this won't actually work and I should completely uninstall ELL and start over?

lovettchris, 2021-02-06:
Ok great, using miniconda is fine, and if "import ell" works then you should be good to go.  You can run this test from your build folder to be sure it is working properly:
```
ctest . --build-config release -R python
```

shalaka188, 2021-03-19:
Hiii ,
even I am facing the same problem
I used anaconda prompt of version 3.18 & as directed i have created an environment py36 & done as directed but still facing issue in
 _ctest . --build-config release -R ell-python-interface-test___

ouput is :

Test project C:/Users/Shalaka.DESKTOP-9R8N143/ELL/build
    Start 44: ell-python-interface-test
1/1 Test #44: ell-python-interface-test ........***Failed    0.06 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.12 sec

The following tests FAILED:
         44 - ell-python-interface-test (Failed)
Errors while running CTest_

but I did,'t install python 3.6 
was that neccessary?

plz help me out 
Thank you..