microsoft/ELL #255: cmake configure failed with could not find TARGET _ELL_python

rqbh, 2020-10-16:
Found LLVM 8.0.1
Using LLVMConfig.cmake in: /usr/lib/llvm-8/cmake
-- Found SWIG_EXECUTABLE=/usr/local/bin/swig
Skipping pitest because one of RPI_CLUSTER, RPI_PASSWORD, RPI_KEY, TEST_MODELS_REPO is missing
Creating wrappers for python
CMake Error at CMake/CommonInterfaces.cmake:177 (set_property):
  set_property could not find TARGET _ELL_python.  Perhaps it has not yet
  been created.
Call Stack (most recent call first):
  CMake/CommonInterfaces.cmake:194 (generate_interface_module)
  interfaces/python/CMakeLists.txt:21 (generate_interface)

lovettchris, 2020-12-02:
If you have not setup a python environment and you do not want the python bindings for ELL then try running "cmake" with the additional flag: -D DISABLE_PYTHON=ON.  There is a small bug in our CMakeLists.txt file that needs to set PYTHON_ENABLED to OFF in this case as shown in this git diff:

```cmake
diff --git a/CMakeLists.txt b/CMakeLists.txt
index d01e464fd..4f164e8cf 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -62,6 +62,7 @@ set(RPI_KEY "$ENV{RPI_APIKEY}")

 if(DISABLE_PYTHON)
   message(STATUS "Python has been explicitly disabled")
+  set(PYTHON_ENABLED OFF)
 else()
   foreach(PYTHON_VERSION 3.7 3.6)
     find_package(PythonInterp ${PYTHON_VERSION} QUIET)
```

If you do want the python bindings then you need to configure your python environment.  This [Dockerfile ](https://github.com/microsoft/ELL/blob/master/scripts/CI/Linux/Dockerfile) shows how to do that.  And this image is published to docker as [cjlovett/ell-dependencies](https://hub.docker.com/repository/docker/cjlovett/ell-dependencies), so you could just use this docker container to build ell.

lovettchris, 2020-12-03:
This is fixed now.

closed 2020-12-03 by lovettchris