microsoft/ELL #256: ctest error : model_test failed, common_test failed

MoonSunBeen, 2021-01-04:
@lovettchris 
Hello, I have the same problem. 

When I enter the cmd `ctest . --build-config release -R ell-python-interface-test -VV`
It is part of my error code 
`44: DoubleVector test ... Passed` 
`44: FloatVector test ... Passed`
`44: vector_test passed`
`44: model_test failed`
`44: exception:`
`44: Loading file ../../../examples/models/model_1.model`
`44: common_test failed`
`44: trainers_test.test -- TBD`
`44: trainers_test passed`
`44: test_tensor ... Passed`
`44: test_predictor ... Passed`

model_test and common_test are failed

So, I've check the requirements.txt
`(py36) C:\Program Files\ELL\interfaces\python\package\ell>python -m pip freeze`
`appdirs==1.4.4`
`audioread==2.1.9`
`bcrypt==3.2.0`
`certifi==2020.12.5`
`cffi==1.14.4`
`chardet==4.0.0`
`cntk==2.7`
`cntk-gpu==2.7`
`configparser==5.0.1`
`cryptography==3.3.1`
`cycler==0.10.0`
`dask==2020.12.0`
`dataclasses==0.7`
`decorator==4.4.2`
`flake8==3.8.4`
`future==0.18.2`
`idna==2.10`
`importlib-metadata==3.3.0`
`iso8601==0.1.13`
`joblib==1.0.0`
`kiwisolver==1.3.1`
`librosa==0.8.0`
`llvmlite==0.35.0`
`matplotlib==3.3.3`
`mccabe==0.6.1`
`numba==0.52.0`
`numpy==1.19.3`
`olefile==0.46`
`onnx==1.8.0`
`opencv-python==4.4.0.46`
`packaging==20.8`
`paramiko==2.7.2`
`parse==1.18.0`
`Pillow @ file:///C:/ci/pillow_1603822370986/work`
`pooch==1.3.0`
`protobuf==3.4.1`
`psutil==5.8.0`
`PyAudio==0.2.11`
`pycodestyle==2.6.0`
`pycparser==2.20`
`pyflakes==2.2.0`
`PyNaCl==1.4.0`
`pyparsing==2.4.7`
`python-dateutil==2.8.1`
`python-speech-features==0.6`
`PyYAML==5.3.1`
`requests==2.25.1`
`resampy==0.2.2`
`scikit-learn==0.24.0`
`scipy==1.5.4`
`serial==0.0.97`
`six @ file:///home/conda/feedstock_root/build_artifacts/six_1590081179328/work`
`SoundFile==0.10.3.post1`
`threadpoolctl==2.1.0`
`torch==1.7.0`
`torchvision==0.8.1`
`typing-extensions @ file:///tmp/build/80754af9/typing_extensions_1598376058250/work`
`urllib3==1.26.2`
`wincertstore==0.2`
`zipp==3.4.0`

I've checked all the package are installed

I have no idea about this problem. Can you suggest me the solution of this problem?

lovettchris, 2021-01-04:
Thanks for filing a new issue.  So looks like "model_test" is failing, but there is no useful diagnostic output.  It looks like it threw an exception, but the exception printing printed nothing, so we only see "exception:".  The test loaded model_1.model but not model_2.model.  So perhaps it didn't find model_1.model.   Can you see "model_1.model" in ~\build\examples\models ?

Can you try running `python model_test.py` from `~\ELL\build\interfaces\python\test` in a debugger and let me know what you see?

Note: it is a bit odd that your ELL repo is under "C:\Program Files\" -- normally this folder does not have user permissions, normally I put my git repos someplace else, liker "c:\git".

MoonSunBeen, 2021-01-05:
`22: Test command: C:\Users\NOTA1204\miniconda3\envs\py36\python.exe "-m" "unittest" "onnx_node_test.py"
22: Test timeout computed to be: 10000000
22: Traceback (most recent call last):
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\runpy.py", line 193, in _run_module_as_main
22:     "__main__", mod_spec)
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\runpy.py", line 85, in _run_code
22:     exec(code, run_globals)
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\unittest\__main__.py", line 18, in <module>
22:     main(module=None)
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\unittest\main.py", line 94, in __init__
22:     self.parseArgs(argv)
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\unittest\main.py", line 141, in parseArgs
22:     self.createTests()
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\unittest\main.py", line 148, in createTests
22:     self.module)
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\unittest\loader.py", line 219, in loadTestsFromNames
22:     suites = [self.loadTestsFromName(name, module) for name in names]
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\unittest\loader.py", line 219, in <listcomp>
22:     suites = [self.loadTestsFromName(name, module) for name in names]
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\unittest\loader.py", line 153, in loadTestsFromName
22:     module = __import__(module_name)
22:   File "C:\Git\ELL\build\tools\importers\onnx\test\onnx_node_test.py", line 15, in <module>
22:     import onnx
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\site-packages\onnx\__init__.py", line 9, in <module>
22:     from onnx.external_data_helper import load_external_data_for_model, write_external_data_tensors
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\site-packages\onnx\external_data_helper.py", line 10, in <module>
22:     from .onnx_pb import TensorProto, ModelProto
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\site-packages\onnx\onnx_pb.py", line 8, in <module>
22:     from .onnx_ml_pb2 import *  # noqa
22:   File "C:\Users\NOTA1204\miniconda3\envs\py36\lib\site-packages\onnx\onnx_ml_pb2.py", line 22, in <module>
22:     create_key=_descriptor._internal_create_key,
22: AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
22/44 Test #22: onnx_importer_test ...............***Failed    0.23 sec
test 23
      Start 23: forestTrainer_test`

MoonSunBeen, 2021-01-05:
`AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'`

Fixed this error 
`pip uninstall protobuf python3-protobuf`
`pip install --upgrade pip`
`pip install --upgrade protobuf`

above these codes

Then ctest passed.

closed 2021-01-05 by MoonSunBeen

lovettchris, 2021-01-05:
Wow, ok, interesting!  Thanks for closing the issue.